prusa3d/Prusa-Firmware #4825: [BUG] 3.14.1 MK3S+ firmware installation failed

A11lessandro2, 2024-12-21:
### Printer model

MK3S/+

### MMU model

MMU3

### Firmware versions

Standard 3.14.1, downloaded from Prusa

### Printing from...

SD Card

### Describe the bug

When I try to install the firmware 3.14.1 on my prusa mk3s+ printer, I get the error installation failed.
The firmware installs anyway, but it doesn't allow me to select the Italian language or any other language,
if I do it brings me back to the language selection menu, you can only select the English language.
By selecting the English language the printer still works and I can print.
At the moment I have reinstalled the old firmware 3.13.2.7080 and that loads without problems.
I know that this problem is an old problem, but it continues to appear at regular intervals, some firmwares work well
like the 3.13.2.7080, while others just give an installation error.

3d-gussner, 2024-12-23:
@A11lessandro2 Good morning. How do you flash your printer?
Please use PrusaSlicer, connect via USB to your printer and follow the instructions https://help.prusa3d.com/article/how-to-update-firmware-mk3s-mk3s-mk3_2227

Often these kind of reported issues are a result that the firmware has been flashed via OctoPrint firmware updater or other ways where the xflash hasn't been updated.

There is a pull request for the OctoPrint updater https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/325 but that has not been merged.

If you still have some issues please use a different USB cable and/or USB port.

3d-gussner, 2025-01-21:
Closing due to no reaction. Feel free to reopen it.

closed 2025-01-21 by 3d-gussner