prusa3d/Prusa-Firmware #4826: [QUESTION] Build of FW 3.14.1 fails

mi3dke, 2024-12-25:
### Question

I am building the firmware for my MK3S+ in VSCode. Though I am not a developer I am also not a rookie (I thought) and build my firmware since version 3.11. 
But during the build of 3.14.1 I get an error. Subsequently the hex file is not build. 
I updated to the latest versions of cmake and python.
Any suggestion or further information to get it back running?

Thanks
Mike

pyth_on .\utils\bo_otstrap.py
Downloading ninja-1.12.1
Extracting ninja-1.12.1
Downloading cmake-3.22.5
Extracting cmake-3.22.5
Downloading avr-gcc-7.3.0
Extracting avr-gcc-7.3.0
Downloading prusa3dboards-1.0.6
Extracting prusa3dboards-1.0.6
Traceback (most recent call last):
  File "C:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware\utils\bootstrap.py", line 207, in <module>
    sys.exit(main())
  File "C:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware\utils\bootstrap.py", line 190, in main       
    install_dependency(dependency)
  File "C:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware\utils\bootstrap.py", line 145, in install_dependency
    download_and_unzip(url=url, directory=installation_directory)
  File "C:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware\utils\bootstrap.py", line 98, in download_and_unzip
    obj.extractall(path=str(extract_dir), filter='data')
TypeError: extractall() got an unexpected keyword argument 'filter'

Subsequently the build fails.
[proc] Executing command: c:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake --version
[proc] Executing command: c:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -E capabilities
[kit] Successfully loaded 0 kits from C:\Users\micha\AppData\Local\CMakeTools\cmake-tools-kits.json
[kit] Successfully loaded 1 kits from c:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware\.vscode\cmake-kits.json
[variant] Loaded new set of variants
[main] Configuring project: Prusa-Firmware 
[proc] Executing command: c:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_TOOLCHAIN_FILE:FILEPATH=c:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware/cmake/AvrGcc.cmake -DCMAKE_MAKE_PROGRAM:STRING=c:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/ninja-1.12.1/ninja --no-warn-unused-cli -SC:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware -Bc:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Found Python3: C:/Users/micha/AppData/Local/Programs/Python/Python310/python.exe (found version "3.10.10") found components: Interpreter 
[cmake] -- Project version (Configuration.h): 3.14.1.64
[cmake] -- Filename suffix..................: FW_3.14.1
[cmake] -- Filename debug suffix ...........: FW_3.14.1+8237
[cmake] -- Git was not found or an error occurred parsing the tag. Falling back to Configuration.h values (3.14.1.64).
[cmake] -- Project version git..............: 3.14.1.64
[cmake] -- Project version git hash.........: HEAD-FORM
[cmake] -- Project version git description..: -128-NOTFOUND--128-NOTFOUND
[cmake] -- Filename suffix .................: FW_3.14.1
[cmake] -- Filename debug suffix ...........: FW_3.14.1+8237
[cmake] ToolChain dir is C:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/avr-gcc-7.3.0
[cmake] ToolChain dir is C:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/avr-gcc-7.3.0
[cmake] -- The C compiler identification is GNU 7.3.0
[cmake] -- The CXX compiler identification is GNU 7.3.0
[cmake] -- Detecting C compiler ABI info
[cmake] ToolChain dir is C:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/avr-gcc-7.3.0
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: C:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/avr-gcc-7.3.0/bin/avr-gcc.exe - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] ToolChain dir is C:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/avr-gcc-7.3.0
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: C:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/avr-gcc-7.3.0/bin/avr-g++.exe - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] Language maximum size (from config.h): 13568 bytes
[cmake] Variant added: MK25-RAMBo10a
[cmake] Variant added: MK25-RAMBo13a
[cmake] Variant added: MK25S-RAMBo10a
[cmake] Variant added: MK25S-RAMBo13a
[cmake] Variant added: MK3-E3DREVO
[cmake] Variant added: MK3-E3DREVO_HF_60W
[cmake] Variant added: MK3
[cmake] Variant added: MK3S-E3DREVO
[cmake] Variant added: MK3S-E3DREVO_HF_60W
[cmake] Variant added: MK3S
[cmake] -- Configuring done
[cmake] CMake Error at CMakeLists.txt:343 (add_library):
[cmake]   Cannot find source file:
[cmake] 
[cmake]     C:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/prusa3dboards-1.0.6/cores/prusa_einsy_rambo/wiring_digital.c
[cmake] 
[cmake]   Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
[cmake]   .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
[cmake] 
[cmake] 
[cmake] CMake Error at CMakeLists.txt:343 (add_library):
[cmake]   No SOURCES given to target: avr_core
[cmake] 
[cmake] 
[cmake] CMake Generate step failed.  Build files cannot be regenerated correctly.
[proc] The command: c:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware/.dependencies/cmake-3.22.5/bin/cmake -DCMAKE_BUILD_TYPE:STRING=Release -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_TOOLCHAIN_FILE:FILEPATH=c:\Users\micha\Arduino\_Prusa\3.14.1\Prusa-Firmware/cmake/AvrGcc.cmake -DCMAKE_MAKE_PROGRAM:STRING=c:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/.dependencies/ninja-1.12.1/ninja --no-warn-unused-cli -SC:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware -Bc:/Users/micha/Arduino/_Prusa/3.14.1/Prusa-Firmware/build -G Ninja exited with code: 1

gudnimg, 2024-12-26:
Its likely the fix for a deprecation warning I made recently doesn't work on Python 3.10.10 🤔 It should work on Python 3.12. 

For now, if you want to use Python 3.10.10, just remove the `filter` parameter. Then it should work.

mi3dke, 2024-12-26:
I uninstalled all previous versions and reinstalled cmake and python.
Works again now. Thanks.

mi3dke, 2024-12-26:
Update frameworks.

closed 2024-12-26 by mi3dke