prusa3d/Prusa-Firmware #4828: [BUG] Axis Dimension wrong

ChopaChupChup, 2024-12-28:
### Printer model

MK3

### MMU model

No-MMU

### Firmware versions

3.14.1

### Printing from...

SD Card

### Describe the bug

Extruder overdrives (try to move 1 Millimeter, instead it does like 20). 
Z-Axis Dimension to small (try to move 100 Millimeter, instead it does like 10)

I have a MK3 and came from 3.10.1 to 3.14.1 directly.
Before update no issues.

ChopaChupChup, 2024-12-28:
Going to an older firmware MK3 version did not help.
I had to update a MK3S firmware, then the bug was gone.
After that I could update the firmware to the new MK3 and the problem was not more present

3d-gussner, 2024-12-30:
@ChopaChupChup Thanks for the report. Do I understand correctly:
- You have a **MK3** and NOT MK3S
- After updating from **MK3** firmware 3.10.1 to **MK3** firmware 3.14.1 you had issues with extruder and motion system
- Updating temporary to **MK3S** 3.14.1 and then back to **MK3** 3.14.1 solved the issue

ChopaChupChup, 2024-12-30:
> @ChopaChupChup Thanks for the report. Do I understand correctly:
>
> - You have a **MK3** and NOT MK3S
>
> - After updating from **MK3** firmware 3.10.1 to **MK3** firmware 3.14.1 you had issues with [...]

Exactly

3d-gussner, 2024-12-30:
That is odd. First idea comes up is that some eeprom has been corrupted.
As your printer is working, do you think is it worth to try to figure out why that may have happened?

If not please close this issue

taraneier, 2025-01-06:
I'm having this issue on my MK3S+ after loading MK3S firmware 3.14.1.  The scale is probably half.

3d-gussner, 2025-01-07:
@taraneier Glad to hear that Andres for Prusa Support could help you and it works again.
As you solved the issue with the factory reset probably some EEPROM vaules got changed. Do you remember any operation where you might have changed the EEPROM values?
- Flash wrong firmware by accident
- Run some GCODE that might have been sliced for a different printer

Please also close this issue as it has been solved.

taraneier, 2025-01-07:
I can't see it being a case of wrong firmware. My problems began with flashing 3.14.1 and I can't remember the last time I updated the firmware. I used the same hex file after the factory reset when flashing 3.14.1.  I also added 3.13.1 and 3.10.? after the problem showed up but before contacting Prusa support.  It was very strange to me that it was changing the operating scale on X and Y by half but doubling the E. Then with diagnostics it threw hardware errors that are not detected after the factory reset. We did tune the x-axis pully and tighten the belt. 

I can't see GCODE either. My prints are either things I've designed in Fusion 360 or stl files from thingiverse or printables. I run them all through Prusa Slicer.

I'm happy to keep sleuthing this out, but I will mark it as closed as mine is now working and we have a path for others to repeat.

taraneier, 2025-01-07:
I think ChopaChupChup will have to close this, it's not an option for me.

closed 2025-01-14 by 3d-gussner