prusa3d/Prusa-Firmware #4830: [BUG] 3.14.1 - self test problem - filament fan error

dturak, 2025-01-01:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.1

### Printing from...

SD Card

### Describe the bug

during the self-test and fan checking, the first hotend fan passes correctly, however, during the filament fan test, a message pops up asking whether the fan is spinning, after confirming that it is, an error pops up and the self-test is interrupted.

I have a REVO hotend but I checked on every version 3.14.1 and on version 3.14.0 and the effect was the same. Only when I installed version 3.13.3 the test passed successfully

### How to reproduce

start self test -> when asked if the fan is spinning, select that it is spinning -> an error appears on the screen

### Expected behavior

when it is confirmed that the fan is tested successfully, the fan test is completed and the next step of the self-test is performed.

3d-gussner, 2025-01-03:
Thanks for the report.
When the part cooling fan test fails you get two questions to be answered correctly:
```
Fan test
Left hotend fan?
 Spinning
>Not spinning
```
after confirming that the left hotend fan is `Not spinning` you get
```
Fan
Front print fan?
 Spinning
>Not spinning
```
Both selections are default set to `Not spinning`

Please ensure that you select on the first question `Not spinning` as the hotend fan passed the test
AND
select `Spinning` for the Front part fan.

I just tried to simulate your issue and the only way I got an error
```
Selftest error!
Please check:
Front/left fans
Swapped
```
when I selected `Spinning` at the first question for the left hotend fan.

3d-gussner, 2025-01-03:
As the fan has been detected correctly in FW 3.13.3 I would suggest to check if the fan still works.
Is it a stock Prusa part cooling fan or do you use a 3rd party fan?

Troubleshoot fans:
- Set in LCD menu Settings -> Temperature -> Fan speed: to 255
- In LCD Support menu -> Extruder info you can see the detected RPM of the fans
- To trouble shoot the hotend fan you can set the hotend temperature above 56 °C in the Settings -> Temperature -> Nozzle menu 

My printer shows
Hotend fan: ~ 3720 - 3900
Print fan: ~ 4440 - 4560

The MK404 SIM
Hotend fan: ~ 3240 - 3300
Print fan: ~4440 - 4740

Don't forget to cool down after troubleshooting.

You can also connect via USB to the printer and use `M155 S1 C2` to show the detected fan RPM and

dturak, 2025-01-12:
after doing all this it started working. Thank you very much. I'm closing the bug

closed 2025-01-12 by dturak