prusa3d/Prusa-Firmware #4835: [BUG]

ftmiranda, 2025-01-17:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.1

### Upgrades and modifications

None

### Printing from...

OctoPrint

### Describe the bug

After the upgrade to this firmware I did a printer reset, delete all data. So brand new calibration and printer setup.
What happens is, when I switch to bed meshing 7x7, which should give me 49 check points, it does for a couple of prints, after that it goes crazy and checks only 21 points, sometimes 12 points, sometimes 29 points... 
I did A reset again, factory reset, calibrated the printer all over, then changed to bed meshing to 7x7, it works again a few times the 49 point checks, then it goes ramdom.

### How to reproduce

I noticed this happens when I send the gcode via octoprint to the printer, and for some reason, the first layer fails and I from the printer select stop printing. 
After cleaning the bed and pre-heating again, I send the same gcode again via octoprint and the ramdom point calibration check starts all over. 
I noticed that if I power down the printer and turn it back on, sometimes it goes back to 49 point checks, sometimes it does not... 
I already engaged support on this, and they told me this looks like a firmware issue, for me to open a case here. 

### Expected behavior

always check 49 points 

### Files

let me know what information would you need to help debug this issue, thank you

3d-gussner, 2025-01-17:
@ftmiranda Thanks for the report. With FW 3.14.0 we introduced Mesh Bed Leveling (MBL) improvements [see](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.0).

After you upgraded the printer or factory reset it and switch to 7x7, the calibration will run a full 7x7 49 point scan.

Older G-code files which use only `G80` to start the MBL will always execute the 3x3 9 or 7x7 49 MBL before the print.
Newer PrusaSlicer versions and profiles will send an extended `G80` command including the print area of the print. This will only scan the bed at the needed locations and so reduce the pre-print MBL sequence.

Example:
- Old G-code of a centered at X125 Y105 cube with 25x25x20 dimensions will have
```
G80 ; mesh bed levelling
```  
  - This will run the full 3x3 or 7x7 MBL
  - Time to run
    - 08:38:42,891 Start
    - 08:39:05,011 End
    - around 22 seconds

[image]

- New G-code of the same centered cube
```
G80 X112.5 Y92.5 W25 H25 ; mesh bed levelling
```
  - When 7x7 MBL is active in the Settings, this will run only run a 3x3 MBL around the print area.
    - Center is X125 and Y105
    - The print without the default skirt
      - starts at X112.5 Y92.5 with the width of 25 and height of 25
        - X125 - X(25/2) = 112.5
        - Y105 - Y(25/2) = 92.5 
    - The MBL will be only performed in the area where the print is. In this case only a 3x3 scan
      - X69 Y69, X103 Y69, X137 Y69
      - X137 Y103, X103 Y103, X69 Y103
      - X69 Y137, X103 Y137, X137 Y137

  - Time to run
    - 08:39:31,449 Start
    - 08:39:36,984 End
    - around 5.5 seconds

[image]

This should explain your issue, please in the check the G-code files how the `G80` command looks like. Is it only `G90` or has it also the new extended X,Y,W,H parameters.

Also please ensure that you have correct settings in OctoPrint see [here](https://github.com/prusa3d/Prusa-Firmware/issues/4807#issuecomment-2459082569)

ftmiranda, 2025-01-17:
Hello This is the GCode that is returning 12 check points (not 9 not 49)
I always printed using Octoprint just to send the Gcode to the MK3S+ via serial connection.
I don´t leverage all features of octoprint per say, I just use it to have access to the webcam streaming and to send gcode to the printer. 

I'm sorry but I'm not as savvy to understand 1/2 of what you said up there lol 
I read the octoprint post and added those codes to the different sections of the GCode on octoprint, as for the octoprint plug in, I see it only works on MK3S/+, since I'm waiting my MK4S upgrade kit, I guess this pluggin will not work when I upgrade this printer to the MK4S?

I'm not printing any gcode from the web, I'm getting the m3f or the STL, opening my PrusaSlicer and I slice all my new prints, creating new gcode (that I guess should have this new code baked in to allow the 49 check points right?) 

[obj_1_DryPod v8_Top_0.2mm_PLA_MK3S_6h3m.zip](https://github.com/user-attachments/files/18456490/obj_1_DryPod.v8_Top_0.2mm_PLA_MK3S_6h3m.zip)

3d-gussner, 2025-01-17:
@ftmiranda Sorry to answer bit too technical.

I inspected your G-Code and there the MBL is `G80 X82.8005 Y84.8936 W100.996 H50.9542 ; mesh bed levelling` 
This means that the print you have is ~101 mm wide and ~51 mm deep/high.
The firmware only runs the MBL for this area and as it is wider than in my example it performs 5 x 3 grid MBL

How many probes are executed depends on the size and position of the actual print area it can vary from 9 to 49 probe points.

Mine was centered and on 25 mm wide and 25 deep/high so the minimum amount of points 9 are used.
In your case the width is higher and therefore the printer measures 5 points on X and 3 rows on Y.

If you print a 250x210 object it will do the 49 probe points.

>since I'm waiting my MK4S upgrade kit

The MINI, MK3.5, MK3.9, MK4, XL firmware had this for some time (using `M555`) and as this is a big time saver 22 sec down to 5 sec (when the print is small) we also implemented it for the MK2.5/S and MK3/S/+ printers.

>I guess this pluggin will not work when I upgrade this printer to the MK4S?

At this moment the MK4 (and other 32-bit) firmware doesn't support this Plug-in. But as long you use OctoPrint with the MK3S please use the Plug-in and settings.

With the MK4S you may want to consider to use PrusaConnect.

Which RPi and WebCam are you using?

ftmiranda, 2025-01-17:
Oh! I see now what you mean! The check points (up to 49) all depends on the size of the object one will print.
If it is a small object, it will only do the point checks in a certain are of the bed, not the entire 49 points...  if that is true, then it makes total sense. When I was printing the MK4S upgrade parts, it would do the 49 point checks all the time, because most of the bed was populated by parts ! aha! 

I'm using a RPi4 4GB and I'm using this camera:  https://www.amazon.com/gp/product/B07BK1QZ2L?psc=1 (Camera Module 75 Degree 5MP OV5647)

ftmiranda, 2025-01-17:
So basically this is not a bug, this is a feature ! lol
Thank you for the highly detailed answers and clarifications ! I hope this not only helps me but others as well, thank you !

closed 2025-01-17 by ftmiranda

3d-gussner, 2025-01-17:
Glad to help and yes it is a feature.