prusa3d/Prusa-Firmware #4837: [BUG] fail build with  TMC_DEBUG

sl1pkn07, 2025-01-19:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

GIT

### Printing from...

SD Card

### Describe the bug

fail build if TMC_DEBUG is enabled

### How to reproduce

uncomment the line 296 in the MK3S.h variant (for example)
start build process

output:

```
[60/329] Building CXX object build_gen/MK3S/CMakeFiles/MK3S_lang_base.dir/__/__/Firmware/ultralcd.cpp.obj
FAILED: build_gen/MK3S/CMakeFiles/MK3S_lang_base.dir/__/__/Firmware/ultralcd.cpp.obj 
/media/COSAS/aplicaciones/Prusa-Firmware/.dependencies/avr-gcc-7.3.0/bin/avr-g++ -DCMAKE_CONTROL -DFW_COMMITNR=8225 -DFW_COMMIT_HASH=\"6a16284da\" -DFW_COMMIT_HASH_LENGTH=9 -DFW_FLAVERSION=1 -DFW_FLAVOR=BETA -DFW_MAJOR=3 -DFW_MINOR=14 -DFW_REPOSITORY=\"sl1pkn07\" -DFW_REVISION=1 -DFW_VARIANT=\"variants/MK3S.h\" -DLANG_MODE=1 -DSOURCE_DATE_EPOCH=\"2025-01-19\" -DSOURCE_TIME_EPOCH=\"09:32:18\" -I/media/COSAS/aplicaciones/Prusa-Firmware/.dependencies/prusa3dboards-1.0.6/cores/prusa_einsy_rambo -I/media/COSAS/aplicaciones/Prusa-Firmware/.dependencies/prusa3dboards-1.0.6/variants/prusa_einsy_rambo -I/home/sl1pkn07/aplicaciones/Prusa-Firmware/Firmware -Os -g -DNDEBUG -std=gnu++1z -Wall -Wextra -Wno-expansion-to-defined -Wsign-compare -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_PRUSA_EINSY_RAMBO -DARDUINO_ARCH_AVR -fno-threadsafe-statics -fno-rtti -fno-exceptions -fno-unwind-tables -ffunction-sections -fdata-sections -flto -fno-fat-lto-objects -frandom-seed=Firmware/ultralcd.cpp -frandom-seed=Firmware/ultralcd.cpp -MD -MT build_gen/MK3S/CMakeFiles/MK3S_lang_base.dir/__/__/Firmware/ultralcd.cpp.obj -MF build_gen/MK3S/CMakeFiles/MK3S_lang_base.dir/__/__/Firmware/ultralcd.cpp.obj.d -o build_gen/MK3S/CMakeFiles/MK3S_lang_base.dir/__/__/Firmware/ultralcd.cpp.obj -c /home/sl1pkn07/aplicaciones/Prusa-Firmware/Firmware/ultralcd.cpp
In file included from /home/sl1pkn07/aplicaciones/Prusa-Firmware/Firmware/ultralcd.cpp:19:0:
/home/sl1pkn07/aplicaciones/Prusa-Firmware/Firmware/ultralcd.cpp: In function 'void lcd_main_menu()':
/home/sl1pkn07/aplicaciones/Prusa-Firmware/Firmware/menu.h:118:76: error: invalid conversion from 'void (*)(uint8_t) {aka void (*)(unsigned char)}' to 'menu_func_t {aka void (*)()}' [-fpermissive]
 #define MENU_ITEM_FUNCTION_P(str, func) do { menu_item_function_P(str, func); } while (0)
                                                                            ^
/home/sl1pkn07/aplicaciones/Prusa-Firmware/Firmware/ultralcd.cpp:5222:5: note: in expansion of macro 'MENU_ITEM_FUNCTION_P'
     MENU_ITEM_FUNCTION_P(PSTR("recover print"), recover_print);
     ^~~~~~~~~~~~~~~~~~~~
/home/sl1pkn07/aplicaciones/Prusa-Firmware/Firmware/menu.h:119:13: note:   initializing argument 2 of 'void menu_item_function_P(const char*, menu_func_t)'
 extern void menu_item_function_P(const char* str, menu_func_t func);
             ^~~~~~~~~~~~~~~~~~~~
```

### Expected behavior

builds OK

3d-gussner, 2025-01-20:
@sl1pkn07 This code is very very old and it probably got broken over time or was only working in a specific development version.

What do you want to do with `TMC2130_DEBUG`  enabled?
It only added menus to test/trigger recovering and power panic see https://github.com/prusa3d/Prusa-Firmware/blob/e10c8d9c5a99291d838c008b220ce5433778b5b7/Firmware/ultralcd.cpp#L5221-L5224

3d-gussner, 2025-01-20:
> test a faulty board not sure if the driver iself or the circuvent traces

Try to use TMC2130_SERVICE_CODES_M910_M918

sl1pkn07, 2025-01-20:
and the M122?

3d-gussner, 2025-01-20:
@sl1pkn07 `M122` isn't supported on MK3S

sl1pkn07, 2025-01-20:
and in mk3?

3d-gussner, 2025-01-20:
Here you can find all [supported Gcodes](https://prusa3d.github.io/Prusa-Firmware-Doc/Prusa-Firmware/group__GCodes.html)
and [here](https://help.prusa3d.com/de/article/prusa-firmware-spezifische-g-code-befehle_112173) and [here](https://reprap.org/wiki/G-code)

3d-gussner, 2025-01-20:
> and in mk3? or in what version is supported?

The difference between the MK3 and MK3S/+ is mainly the Filament Sensor (MK3 = PAT9125 "laser" and MK3S = IR sensor)

sl1pkn07, 2025-03-07:
then, how i can debug the communication between mcu and the tmc driver if the M122 not exist?

3d-gussner, 2025-03-10:
> M122

Yes you can try to flash vanilla marlin 2.x and find a good MK3S configuration file to trouble shoot your board.

closed 2025-06-17 by github-actions[bot]