prusa3d/Prusa-Firmware #4838: [BUG] After powering on, Z axis moves up twice during filament change

degroof, 2025-01-19:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.1

### Upgrades and modifications

none

### Printing from...

SD Card

### Describe the bug

Minor annoyance: After turning on the printer, the Z axis moves up twice while I'm changing the filament: once when unloading, once when loading. The second one isn't really necessary, and can result in the Z axis topping out if it's already near the top.

Again, not a big deal but hopefully an easy fix. And it only seems to happen after turning the power on. I thought performing an auto-home after startup would be a viable workaround (assumption: firmware doesn't know actual Z height until homed) but it does it in that case too.

### How to reproduce

1. Power on the printer
2. Select "unload filament" (printhead heats up)
3. Press knob to eject filament (printhead moves up and ejects filament)
4. Insert new filament (printhead pulls in filament, then moves up _again_)

### Expected behavior

1. Power on the printer
2. Select "unload filament" (printhead heats up)
3. Press knob to eject filament (printhead moves up and ejects filament)
4. Insert new filament (printhead pulls in filament, with no additional Z movement)

sarusani, 2025-01-24:
The configured minimum raise for unload is 20mm, for load it's 50mm. Therefore the head moves up an additional 30mm.

I will change the behavior to align both values (to 35mm) for the next release.

degroof, 2025-01-24:
Ah! Is _that_ what it's doing? Makes sense now. Thanks for the explanation.

PR #4839 (2025-02-11): Align MIN_Z raise for unload & load
Align values for MIN_Z raise, to prevent movement between unload & load operations.

Fixes #4838
closed 2025-02-11 by 3d-gussner