prusa3d/Prusa-Firmware #4842: [BUG]

mhouse1, 2025-02-05:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.1

### Upgrades and modifications

revo six

### Printing from...

SD Card

### Describe the bug

nozzle diameter  changes randomly after a print (seems to be 5% of the time while using SD card), I suspect this is happening because the nozzle  diameter menu is directly below  the "Print From SD" menu and only requires one click to change nozzle diameter. This causes the next print to warn about nozzle diameter being wrong.

Can we  move  the nozzle diameter menu below preheat menu? I think users use preheat menu more frequently than nozzle diameter menu anyways.

### How to reproduce

Keep using the "Print From SD" menu

3d-gussner, 2025-02-10:
@mhouse1 As the REVO firmware is a community based firmware, the REVO users should decide/discuss where the Quick Nozzle change should be placed on the LCD menu.

3d-gussner, 2025-02-10:
The REVO LCD menu at this moment looks like this
- Info screen
- Preheat
- Print form SD
- Sheet [name] (when multiple sheets are active)
- Nozzle [size]
- AutoLoad filament / Load Filament (depending on settings)

The request from the REVO community was to get the nozzle size into the 1 layer of the LCD menu.
Following your argument also the Sheet menu should be moved as one "accidental" click would change the sheet type.

But where should we place it? An accidental click on AutoLoad / Load / Unlod filament also triggers a function.

mhouse1, 2025-02-11:
I would propose moving Preheat below Print From SD.
For example:
- Info screen
- Print from SD
- Preheat
- Sheet [name] (when multiple sheets are active)
- Nozzle [size]
- AutoLoad filament / Load Filament (depending on settings)

or make it similar to sheet name where it would allows turning on/off multiple nozzle option; i'd think nozzle changes are much less frequent than sheet name anyway.

closed 2025-06-17 by github-actions[bot]