prusa3d/Prusa-Firmware #4855: 2.9.1 - Sequential Printing should not rehome if a crash or other issue is detected.

thew1se, 2025-03-31:
### Description of the bug

During a sequential print, my printed decideded to rehome, which caused the print head to crash into previously completed parts, and damage the parts.   After rehoming, it didn't even return to the correct location to continue printing.  Could have saved 1 part.

MK3S+
Firmware 3.13.3-7094  (Not updated purposely because of this issue which was never addressed - [Issue 3563](https://github.com/prusa3d/Prusa-Firmware/issues/3563#issuecomment-2661791059) )
Prusa Slicer 2.9.1

[image]

[image]

### Project file & How to reproduce

[Rocket.zip](https://github.com/user-attachments/files/19529019/Rocket.zip)

### Checklist of files included above

- [x] Project file
- [x] Screenshot

### Version of PrusaSlicer

2.9.1

### Operating system

Windows 11

### Printer model

MK3S+

3d-gussner, 2025-03-31:
@thew1se Please read https://github.com/prusa3d/Prusa-Firmware/issues/3623

Adding the changes to PrusaSlicer Profile helps with that issue you described.

>Firmware 3.13.3-7094 (Not updated purposely because of this issue which was never addressed - https://github.com/prusa3d/Prusa-Firmware/issues/3563#issuecomment-2661791059 )

Did you try FW 3.14.1? What exactly happens?

>2.9.1 - Sequential Printing should not rehome if a crash or other issue is detected.

The re-home of X and Y attempt is working good/well when a crash is detected and saves a lot of prints. With the default PrusaSlicer profiles the sequential prints may fail.

3d-gussner, 2025-03-31:
I ran several G80  and G81 to see if I can reproduce the MBL issue. FW 3.14.1 everything original black Nuctua and Prusa 5V fan
- No fans ✔ 
- hotend fan on ✔ 
- part fan 100% ✔ 
- hotend and part fan 100% ✔ 
- part fan 50% `M106 S125` with RPM ~3000 - 3060 PrusaFan ✔

3d-gussner, 2025-04-03:
@thew1se Sorry that is used the Gcodes trying to explain my test.
`G80` starts the Mesh Bed Leveling and is also added by PrusaSlicer automatically.
`G81` shows the MBL map. I used that to see if the differences between the tests are outside what can be expected.

>Do these tests you have done address what I am saying about rehoming after crash

Yes I tested the MBL issue on my real/physical MK3S+.
We also use very often the [MK404 SIM](https://github.com/vintagepc/MK404) for testing and as this passed the test before I wanted to ensure that I get same results on a real printer.

>Also the reference to issue 3623 does not provide any other instructions.

Here a link to my comment https://github.com/prusa3d/Prusa-Firmware/issues/3496#issuecomment-2598320727.

With the newest firmware and these changes in your PrusaSlicer profile you should be able to prevent these issues during sequential printing.

Can't tell if and/or if these changes will ever been added to PrusaSlicer profiles. PrusaSlicer improved several things for sequential printing lately.

closed 2025-06-10 by github-actions[bot]