prusa3d/Prusa-Firmware #4861: [BUG] Power Panic with Prusa Connect over USB causes Z shift

TheClifff, 2025-04-28:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.1-8237

### Upgrades and modifications

None

### Printing from...

PrusaConnect

### Describe the bug

When Power Panic occurs and the Printer is connected to Prusa Connect via a USB cable to a Raspberry Pi 3b+ (which also loses power) the Printer will try to recover the print. The printer will move up from the location where the power outage occurs (so far so good). 

Then when the Pi comes online and "reboots" the printer it moves the Z axis up again by 20mm or so, on top of what it did when it initially recovered.

When the printer then says "ready to recover" it recovers from this higher position and not the correct position which ends up ruining the print. 

### How to reproduce

Cut power to both the Printer and Pi during a print 

### Expected behavior

I expect the Printer to move down that distance again or to not move it up in the first place since it only does this as part of the power panic recovery and not when the printer normally boots.

3d-gussner, 2025-04-29:
@TheClifff Which PrusaLink version do you have installed on your RPi 3+?

I have just tested again FW 3.14.1-8237 with PrusaLink 0.8.1 with the MK404 simulator and PrusaLink running on my Ubuntu
1. Start MK404 with Prusa_MK3S
2. Start PrusaLink 0.8.1
3. Start print from PrusaConnect (PrusaLink gcode)
4. Trigger Power Panic
5. Kill PrusaLink on Ubuntu to simulate a crashed RPi with PrusaLink
6. Printer starts up and goes into Pause with a lift of additional 20 and heats the bed
7. Waited few seconds with the start of Prusalink (simulate boot time of RPi)
8. Wait until PrusaLink is back online and wait for LCD message that the print can be recovered
9. In LCD menu Resume print
10. Printer heat up the hotend to previous temperature. Still at same Z height
11. Print recovers at the previous Z height with the print.

Please ensure that you have PrusaLink 0.8.1 installed as the new power panic feature for USB prints only works with FW 3.14.1 + PL 0.8.1 (or higher)

If there are multiple PowerPanic events one after another the recovery may fail.

nokedli172, 2025-12-07:
Any updates on this? I'm still having the same issue with latest MK3S firmware 3.14.1-8237 and PrusaLink 0.8.1 on a Raspberry Pi 3B+. After a power outage, the printer first tries to recover the print on its own while the Pi is still booting. It goes to the exact point it left off but pauses immediately and raises the Z-axis. Then, once the Pi has booted it resets the printer and initiates print recovery. However it starts at a higher Z-height.