prusa3d/Prusa-Firmware #4868: [BUG] mk3s wizard calibration selection of filament before z value calibration

Kubiskoi, 2025-06-18:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.1

### Printing from...

SD Card

### Describe the bug

During wizard calibration before z value calibration printer asks if filament is loaded, when the answer is No, it will ask what type of filament user want to load but the screen is blank (it shines blue but no white text) then when user just turns the knob the text appears and user can choose PLA,PTEG etc.

User can have a feeling that the calibration is broken but it is just visual bug.

### How to reproduce

Run the calibration wizard without loaded filament and it should happen before z value calibration.

### Expected behavior

Options of filament types should be visible immediately

Kubiskoi, 2025-06-19:
1. Unload the filament
2. Go to Calibration > Wizard and follow the wizard steps, the printer will ask what type of filament you want to load but the screen will be empty
3. when the screen is empty don't press the knob but turn it and PLA,PTEG will display on the screen

3d-gussner, 2025-06-19:
@Kubiskoi I just ran a factory reset wizard on latest development state (MK3 branch) and saw that the LCD screen is indeed "clear" not showing the preheat menu until you turn the knob. I will look into it.