prusa3d/Prusa-Firmware #4872: [BUG] Z axis should move upward only once; it moved up and up and up until crash

Fojtik, 2025-07-01:
### Printer model

MK3S/+

### MMU model

No-MMU

### Firmware versions

3.14.1

### Printing from...

SD Card

### Describe the bug

Every operation cauzes Z axis to move up until crash to limit.

### How to reproduce

The problem is when you:
1, load filament;
2 unload filament 
3, load another filament. 

### Expected behavior

Z Axis should move up only once.

3d-gussner, 2025-07-01:
@Fojtik I can't reproduce your issue.
1. Power up MK3S FW3.14.1
2. LCD shows `0.15?` Z height as it is unknown to the printer
3. LCD Load filament moves the Z axis to `50.00?`
4. LCD Unload filament doesn't move the Z Axis
5. Repeated the Steps 3 and 4 multiple times and it just doesn't raise the Z axis another time

Can you write the steps you did to get the multiple raises?

3d-gussner, 2025-07-02:
@Fojtik I can confirm when the filament is loaded, then the printer is powered up and you unload filament it raises by 20 and to 50 when loading.

I have to look into the current development version if we already fixed that issue.