prusa3d/Prusa-Firmware #4874: [BUG]

neoyarus, 2025-07-10:
### Printer model

MK3S/+

### MMU model

MMU3

### Firmware versions

3.14.1

### Upgrades and modifications

RPI Zero 2w attached to the main board running prusalink 0.8.1, usb camera connected, one extra frint fan added.

### Printing from...

PrusaConnect

### Describe the bug

Whenever I finish a print with the MMU, PrusaConnect stops seeing the printer as a MK3S+ with an MMU3 and starts seeing it as a regular MK3S+. This then prevents me from uploading MMU gcodes to the printer from PrusaSlicer since it is now considered incompatible. Resetting or power cycling the printer does not help. If I go into PrusaLink and unlink the printer from PrusaConnect, then link in again, it comes back as MK3S+ with MMU3, but will revert to a regular MK3S+ after a print.

### How to reproduce

Simply run any MMU print on a MK3S+ equipped with an MMU3, and check its status in PrusaConnect after the print finishes.

3d-gussner, 2025-07-10:
@neoyarus I assume you have an issue with PrusaLink see [here](https://github.com/prusa3d/Prusa-Link/pull/1009) Please contact PrusaSupport or send me an email.

neoyarus, 2025-07-12:
> @neoyarus I assume you have an issue with PrusaLink see [here](https://github.com/prusa3d/Prusa-Link/pull/1009) Please contact PrusaSupport or send me an email.

Applied the fix from your pr, it seems to have fixed the issue, thanks!

closed 2025-07-12 by neoyarus