prusa3d/Prusa-Firmware #4875: [FEATURE REQUEST]  perimeters extrusion reduction on overhangs

yoshimitsuspeed, 2025-07-11:
### Printer model

- [ ] MK3S/+
- [ ] MK3
- [ ] MK2.5S
- [ ] MK2.5

### Describe the feature

Over extrusion plays a big part in increasing upward curling with steeper overhang angles. 

Slight over extrusion or maximized extrusion results in the strongest parts, least voids, etc. 

This often creates a challenge needing to balance extrusion to where overhangs don't curl up faster than the build rate, sometimes sacrificing slight under extrusion in other areas. 

I would imagine something similar to dynamic overhang speed where you could reduce flow by a given percent at a given angle or overlap. 
It does get a little tricky on the external perimeter as you probably want it to go back to 100% flow before 0 overlap and for bridges. 
So for example you could do something like 100% flow for 0 overlap, 98% flow for 25% overlap, 99.5% flow for 50% and 100% for 75% overlap.  
Even more control might be to control external perimeters and perimeters separately but that may not be necessary. 

This could be especially useful for TPU and other materials that really like to curl. It would of course sacrifice some geometric accuracy putting less material there but you are doing that anyway once your curl gets too big.

3d-gussner, 2025-07-11:
@yoshimitsuspeed This looks like more to be a request for the PrusaSlicer.

yoshimitsuspeed, 2025-07-11:
> [@yoshimitsuspeed](https://github.com/yoshimitsuspeed) This looks like more to be a request for the PrusaSlicer.

Ugh yes it is. I didn't notice this wasn't the slicer page.

3d-gussner, 2025-07-14:
@yoshimitsuspeed Please open the ticket in https://github.com/prusa3d/PrusaSlicer 
Closing it here

closed 2025-07-14 by 3d-gussner