prusa3d/Prusa-Firmware #4876: [BUG]

Honza51, 2025-07-15:
### Printer model

MK3S/+

### MMU model

MMU3

### Firmware versions

3.14.1 and MMU is 3.0.3

### Printing from...

SD Card

### Describe the bug

When using MMU3 in single mode, if a filament change fails (e.g. M600), the printer pauses correctly and allows manual filament fix (cut blob at filament end). After resuming, the nozzle prints “in the air”—Z does not return to the correct height. It appears that Z‑height babystep or mesh leveling offsets are lost or not re‑applied.

### How to reproduce

Start a print with MMU3 in single mode.

Trigger filament change (M600) 

Not load the filament (try empty position at MMU or filament with bad tip)

Fix/load filament after error message.

Resume print.

Observe nozzle printing above previous layer (missing Z‑offset), it do not lower the z position.

### Expected behavior

After resuming, Z should restore exactly to the previous layer height, with babystep or mesh leveling offsets preserved.

[Chyba 1.zip](https://github.com/user-attachments/files/21233243/Chyba.1.zip)

3d-gussner, 2025-07-16:
@Honza51 I will have a look into that.

Honza51, 2025-07-18:
Thanks! Hope this helps the Prusa community. Seems like others had the same issue on the forums.

gudnimg, 2025-07-18:
>Thanks! Hope this helps the Prusa community. Seems like others had the same issue on the forums.

@Honza51 Any chance you can share a link to the forum discussion?

Honza51, 2025-07-18:
I found several posts related to this issue. Unfortunately, I wasn’t able to locate the first one I originally read, but I’m sharing others that describe the same or very similar behavior.

🔗 https://github.com/prusa3d/Prusa-Firmware/issues/4504 – This issue also reports a Z-height problem after a failed filament load in MMU mode, although in this case the Z-axis error appears to be in the opposite direction (Z moves too low instead of too high).